jneilliii/OctoPrint-TabOrder #1: Can not sort all tabs

EddyMI3d, 2018-01-23:
Hello!
Neat idea with the tab sorting, but I only can sort the "build in" tabs, not the ones that were added with additional plugins, like Print History, Enclosure Plugin, Printer Stats...

Also the subtitle of the Tab Order plugin in the plugin list is a bit weird:
[http://www.eddymi3d.de/Images/OP_TO.jpg](url)

zoombahh, 2018-01-23:
are you sure about this? i can reoder plugin tabs just fine. but you need to find the correct string, as it would appear in the config.yaml file if edited manually.

it seems plugin-tabs are prefixed with "plugin_". for me adding "plugin_printhistory" to the order worked a treat.

[image]

EddyMI3d, 2018-01-23:
Yes, I'm sure - otherwise I would not have written here ...

jneilliii, 2018-01-23:
Which specific plugin are you trying to sort?  Like @zoombahh stated, you can sort plugin tabs, you just have to know the right string to use.

jneilliii, 2018-01-24:
@EddyMI3d, did you CTRL+REFRESH to reload the local cache?  I think that might be your issue.  Like @zoombahh I just installed the print history tab and the string he used for that works perfect for me as well.

closed 2018-01-24 by jneilliii

EddyMI3d, 2018-01-24:
I did CTRL+REFRESH, I did restart OctoPrint, I did a complete reset...

I assume no had a look on the picture I posted (somehow I can't include images in here)
Also the subtitle of the Tab Order plugin in the plugin list is still a bit weird:

[image]

zoombahh, 2018-01-24:
I did look at your image, but that was just the plugin description which jneilliii changed to
"Simple plugin to allow reordering of tabs within OctoPrint" in commit no 0.2.0 ( 8b46181a3b7de1e1933c584ae51a64ce26d774b8 )

maybe you could post a screenshot of your configuration ? the only pitfall ive run in to so far was "ctrl+r" does not clear browser cache, ctrl+f5 and ctrl+<click on refresh button> did clear the cache for me (using chromium, under windows)

if nothing else helps you could try opening debug tools (F12), going to the "network" tab and checking the "disable cache" checkbox, then reload the octoprint page

EddyMI3d, 2018-01-24:
At least CRTL+F5 did it.
Thanks alot!

jneilliii, 2018-01-24:
I'll update the pop up message to include the instructions for CTRL+F5. Release 0.2.0 fixed the plugin manager description.

xhorntail, 2018-01-26:
As I sheepishly posted the same question and was referred to the only other existing question on this project, I figured I'd contribute my findings.

Everything works as expected, to get the proper name for the tabs, simply click them and grab everything after 'http://server.address/#tab_' in your address bar and use that. Be sure not to include the 'tab_' part though.