jneilliii/OctoPrint-TabOrder #4: Not changing tab order with additional tab plugins

Llama-Trauma, 2018-01-31:
Hey there!

I added the plugin, but there were not instructions on how to manage the "additional" tabs I have from new plugin such as a dedicated webcam tab, and the STL Viewer. 

Based on contextual clues, I saw that spaces and capital letters were ignored. I used the + Add Tab Identifier button to add "webcam" and "stlviewer" but that did not enable movement of those tabs with the included arrows in settings. Additionally, I was unable to move any of my tabs using this plugin. I had not tested it prior to installing other plugins, so additional testing is needed, but core functionality was missing either way. 

Appreciate any support or a new release with notes on how to add new plugin tabs please!

jneilliii, 2018-01-31:
Prepend plugin identifiers with plugin_ for example plugin_webcamtab.

closed 2018-01-31 by jneilliii

Llama-Trauma, 2018-02-01:
[image]

[image]

I've just tried your suggestion, saved, refreshed, and even restarted Octoprint - the webcam tab won't move. The "Terminal" tab (just off-screen in the first image) also did not move accordingly. 

Any other tips or tricks?

jneilliii, 2018-02-01:
When you refreshed, did you use ctrl+f5 to cheat the browser cache.

jneilliii, 2018-02-01:
And you have the identifier wrong. It should be plugin_webcamtab. Look at the read me for a better description of getting that identifier.

Llama-Trauma, 2018-02-01:
Yes, I even manually cleared cache.

Llama-Trauma, 2018-02-01:
That worked, thanks for the help! (Sorry for being n00b, and not reading
enough)

jneilliii, 2018-02-01:
No, problem.  I just added the information to the readme a couple of days ago to try and clear up some of the confusion.  If you aren't a plugin developer you may not even know about plugin identifiers.