jneilliii/OctoPrint-TabOrder #15: Taborder never work for me, Yes i have clear the browser cache

ChiliApple, 2018-05-12:
Hy,

the taporder never works for me, since weeks, so i had unstall it. now today with the latest version ido it again, but it doesnt work. the thing is, the new ICONS works look @ screenshot

thx

best Udo
[image]

ChiliApple, 2018-05-12:
[image]

jneilliii, 2018-05-12:
Any errors in browser console or octoprint.log? What version of octoprint and on what os? Check your config.yaml, does the appearance section update at all? If you manually update config.yaml appearance section does it rearrange the tabs?

ChiliApple, 2018-05-12:
no, also manually update config.yaml doesnt work.

[image]

ChiliApple, 2018-05-12:
maybe in the Log the Warnings 
[image]

ChiliApple, 2018-05-12:
i have deinstall the Themeify Plugin, now it works ...

closed 2018-05-12 by ChiliApple

jneilliii, 2018-05-12:
That's strange.  I don't think there is a conflict, but to verify I just installed the Themeify plugin and it is still working for me.  I bet what happened is the uninstall triggered a restart/reload process and so the updated settings in config.yaml picked up.  I'd recommend reinstalling Themeify since I don't think there is an issue there.  If there is, let me know what settings you are using in that plugin.