jneilliii/OctoPrint-TabOrder #18: Strange Behaviour

ygtc, 2018-09-28:
Hi, i like this plugin and i am always using.In these days i noticed strange behaviour.

When i open octoprint or restart octoprint taborder seen like photo1.

[image]

After i click wrench icon top-right corner(config) then close,taborder start works normally again.

[image]

Is that a problem or glitch maybe bug :D 

I just noticed that and i want to ask

jneilliii, 2018-09-28:
Yeah, I think that has to do with the underlying bootstrap tabs.  OctoPrint core programming is handling the stacking of those tabs under the top right list on your initial load.  Not sure if there is a way for me to work around that, but I'll see if I can figure something out.

ygtc, 2018-09-28:
maybe it is only about processing power? I mean maybe orange pi which is better hardware than raspi3 doesnt see this behaviour.I don't know actually if anything i can to help let me know.

ygtc, 2018-09-28:
and also i can share other my observations about this plugin. Like order issue and other plugins corrupt taborder's stable working

jneilliii, 2019-01-01:
So I'm still unable to reproduce this issue on my end.  @ygtc  or @otherguy, could you please provide a screenshot of your Tab Order settings as well as the plugins you have installed/enabled in OctoPrint? Also the exact steps to reproduce the behavior,

jneilliii, 2019-01-01:
Thanks @otherguy, I'm able to reproduce with those steps of a reduced window browser. I'm going to have to look through octoprint's code to see which element I need to trigger the resize action on, because it's a specific element that is doing that stacking in bootstrap I would suspect.

otherguy, 2019-01-01:
I suspect bootstrap handles that. `window.trigger(“resize”)` (pseudocode) should be enough.

jneilliii, 2019-01-01:
Yeah, I ended up testing and using the jQuery resize event seems to work from within developer console.  I'm testing an update to the plugin now to see when to trigger that to resolve.  Will have an updated version by the end of the day probably.

jneilliii, 2019-01-01:
Got it first try.  Seems to be working for me fine now, so I pushed an updated version 0.5.1.  Force check in software update and you should be able to get it.

closed 2019-01-01 by jneilliii

otherguy, 2019-01-02:
Great! Looks good and works just fine 👍 Thanks for fixing it 🙂 

I look at it mostly on my 13" screen -- if you use a big display that explains why you didn't notice it.

Not to push my luck but would you also consider https://github.com/jneilliii/OctoPrint-TabOrder/issues/21 ?

Thanks and have a happy new year!

ygtc, 2019-01-04:
Problem solved for me also!! Thanks for fixing it :smile: 

Keep up to great work m8!! :+1:

PR #20 (2019-01-01): 0.5.1
- Added resize event trigger onAllBound to remove tab stacking when only icons are used. Resolves #18.