jneilliii/OctoPrint-TabOrder #19: Not making new tabs

Lrdnykon, 2018-11-17:
Just installed this on a new octopi 0.15.1 instance on a pi 3 b+ and while I can reorder the 5 tabs that are there by default it wont create new tabs, the other tabs that would be there if there were more than 5 are in the drop down menu to the right.
[image]
[image]

Lrdnykon, 2018-11-17:
What I was trying to say was that I didn't want the drop down menu to be there. I want all the options to be under individual tabs. So if everything was the way I wanted i would have 8 separate tabs. It pops up with all separate tabs when I reload the page but then it moves the other 3 into the drop down menu. I've seen octopi videos where there are more than 5 tabs but mine wont give me that.

jneilliii, 2018-11-17:
Gotcha. My plugin doesn't control that factor, but if you assign an icon to each of the tabs and disable label option it will allow all the tabs to be visible without collapsing I think. The videos you saw may have had a Themify hack applied.