jneilliii/OctoPrint-TabOrder #21: Update to Font Awesome 5

otherguy, 2019-01-02:
Would you consider updating to [Font Awesome 5](https://fontawesome.com/start)? They have a ton of new icons that would be perfectly suited for the tabs!

jneilliii, 2019-01-02:
Thanks for the push. Was holding back on this due to the change in base class names (fa becoming fas) but figured I could make that part of the icon field as well and not hard-code the base class. It will attempt to add the base class to the configured icons, but potentially can break some icons and will have to re-configured in settings.

closed 2019-01-02 by jneilliii

otherguy, 2019-01-02:
Awesome, thanks 😄 Works great, even with the `fab` and `far` classes. I love it! `fab-youtube` for the live streaming plugin is perfect!

Thank you again and I hope you have a great new year!

PR #22 (2019-01-02): 0.5.2
Updated to [fontawesom 5.6.3](https://fontawesome.com/v5.6.3/icons?d=gallery&m=free). As a result, some icons may become broken.  In order to handle the new solid versus branded icons the base icon class (**fas** or **fab**) must also be included in the icon field. Resolves #21.