jneilliii/OctoPrint-TabOrder #25: v0.5.3 makes Temperature → Offset icons disappear

xoan, 2019-01-22:
[image]

jneilliii, 2019-01-27:
I'm not seeing the same thing in my install of version 0.5.3.  It actually should have fixed this issue introduced with version 0.5.2. 

[image]

jneilliii, 2019-01-27:
Do you have touchui plugin installed? I just noticed on my other pi that has that installed I'm seeing this button issue,.

xoan, 2019-01-27:
My apologies. I have Touch UI installed and the issue doesn't appear when Touch UI is disabled.

closed 2019-01-27 by xoan