jneilliii/OctoPrint-TabOrder #27: Icons not showing up

jkkato, 2019-10-07:
I am trying to set up the Tab Order. When I add icons only a square shows up. I even copied the text in the example photos and did the <CTRL> + F5 refresh. I can turn the label on and off and the tool tip and color work.

[image]
[image]

jkkato, 2019-10-07:
I should add that this is the same for both my printers...

[image]

jneilliii, 2019-10-07:
Pet the changelog, 

```
Updated to fontawesom 5.6.3. As a result, some icons may become broken. In order to handle the new solid versus branded icons the base icon class (fas or fab) must also be included in the icon field.
```

jkkato, 2019-10-07:
I did see that changelog, and added the three letters, but it did not work either. One problem is that the instructions does not say how to add these letters to the icon field, so I guessed.

jkkato, 2019-10-07:
I am not at the computer right now... I will be home later.

jneilliii, 2019-10-07:
Here's a quick example.

[image]

jkkato, 2019-10-07:
OK. Thanks. I will try it when I get home. I had put something like "fas-thermometer".

jkkato, 2019-10-07:
Working now. Thanks.

closed 2019-10-07 by jkkato