jneilliii/OctoPrint-TabOrder #28: Icons

leoheck, 2019-10-27:
Hi, where I can find a list of icons?
I saw many pictures with icons, but not a list of them. 

Can you please, share a link with this list or add it in README.md?

Or even better, why this plugin does not come with icons filled by default?
I mean, if the user wants to change the text by and icon it clicks a checkbox, to enable an input field with a default icon there that can be changed too.

What do you think?

jneilliii, 2019-10-28:
Pretty sure there's a link on the readme to fontawesome as well s clicking the info icon.

leoheck, 2019-10-28:
The info icon was good but I had to take some time to understand the syntax, it is not straight forward especially if you don't know what to do. A note on readme will be good. I was not able to find the info on the readme. If it is under changelog it won't work since I was looking for info about the usage, not the changelog.

Well, this is my suggestion to improve it. Have a nice day.

jneilliii, 2019-10-28:
Understood.  I was trying to implement an icon selector option but haven't had much time to work on it.

jneilliii, 2019-11-02:
If you don't mind testing for me that would be great.  You can install a test version by using the url below in plugin manager.  Let me know how it works out.

https://github.com/jneilliii/OctoPrint-TabOrder/archive/0.5.5.zip

leoheck, 2019-11-02:
I am doing it right now.

leoheck, 2019-11-02:
It looks promising dude!
[image]

leoheck, 2019-11-02:
I was able to replace all of my icons. I was using this version that starts with `fa` and they were replaced by something that starts with `fas`
I don't know what this tag does but everything was set just easy. Dude, this is awesome.
[image]

jneilliii, 2019-11-02:
Great, glad it works as expected.  I'll update the version and push an official release.

leoheck, 2019-11-02:
Good. Thanks.
--
Leandro Heck

closed 2019-11-06 by leoheck