jneilliii/OctoPrint-TabOrder #30: Page crashes on iPad with 0.5.5

falvesjr, 2019-11-07:
Love this plugin, but after updating to 0.5.5 from 0.5.4, the OctoPrint page crashes after 10 or 15 seconds in both Chrome and Safari on the iPad. The only update I did today was from 0.5.4 to 0.5.5, so I can only assume that this problem has to do with OP-TO. I have cleared the browser cache and it hasn't helped.

Can I downgrade back to 0.5.4 while you figure this out? If so, where can I get the older version?

Thanks,
Fernando
[image]
[image]

jneilliii, 2019-11-07:
You can install the previous version using the url below in plugin manager. 

https://github.com/jneilliii/OctoPrint-TabOrder/archive/0.5.4.zip

Does the chrome in iPad have a developer console? If so, could you please check for errors there?

falvesjr, 2019-11-07:
Thanks for the link to 0.5.4. I installed it, but the problem is still there in iOS… Does the update to 0.5.5 update any other components that don’t get downgraded if you re-install an older version? I’m not having any issues with browsers on the Mac, but I often use my iPad to check on prints and would hate to lose that ability.

I’ll attach the octoprint logs. I don’t know if they will help, but I guess they won’t hurt! =)

I don’t see any way to access a developer console in Chrome for iOS, so if it’s there, I don’t know how to get to it.

The problem is very weird, Chrome is the most sensitive to it, giving me a page error the very first time it happens. Safari seems to be a little more tolerant as it reloads the page at least once and keeps going, but will usually fail when it tries to reload again. Safari also seems to be reloading a page at octopi.local/#temp.

I hope you can figure this out!

Cheers,
Fernando

jneilliii, 2019-11-07:
I didn't get the log file, but the update doesn't add anything that would still be there after downgrading.  Make sure you've cleared your cache and reloaded the page completely after downgrading.  The only thing that was added in the newer version was the fontawesome icon picker, so I assume it has something to do with that. I have an iPad at the office that I can experiment with to see if I can reproduce the issue.

falvesjr, 2019-11-07:
[octoprint.log](https://github.com/jneilliii/OctoPrint-TabOrder/files/3820352/octoprint.log)
[octoprint.2019-11-04.log](https://github.com/jneilliii/OctoPrint-TabOrder/files/3820353/octoprint.2019-11-04.log)

I guess files don't make it through an email reply, so I've attached them here via the web interface. Please let me know what you find and I'll keep trying to narrow down the problem. Thanks!

falvesjr, 2019-11-07:
I'm in the middle of a print right now, but when it's done later tonight, I'll try restarting the RasPi running OctoPrint as I have been unable to make it work again on my iPad... Maybe the OctoPi server is caching something that keeps it not working on the iPad? I'll let you know...

jneilliii, 2019-11-07:
Yeah, so there's definitely nothing in the log file that indicates an issue, and I just tried in safari on the iPad at work and seems to be fine, no errors or crashes.  I couldn't get Chrome to work because of the way I have SSL certificates installed on my instance and couldn't get the server cert installed. One question though, you mentioned it was after installing the update to Tab Order but I'm wondering if it might be an issue with one of the other plugins, that the upgrade caused some kind of conflict.  If you disable all other plugins does the same thing happen?

falvesjr, 2019-11-08:
I'm starting too have the same feeling that this issue is not related to TabOrder... I just noticed that everything works fine with 0.5.4 if OctoPrint is not printing, making me think this might need related to OctoPrint-DisplayLayerProgress which also updated a couple of days ago. I need a little more testing to confirm this, but if it turns out that it's another plugin causing issues, I want to sincerely apologise for wasting your time!

Are you using OctoPrint-DisplayLayerProgress?

I'll get back to you as soon as I find out more. Cheers!

falvesjr, 2019-11-08:
Well, upon further testing, I'm pretty sure the problem is indeed caused by OctoPrint-DisplayLayerProgress and NOT by TabOrder! I downgraded OctoPrint-DisplayLayerProgress to it's previous version and the problem has disappeared.

Again, I'm sorry to have bothered you with this as I fell victim to coincidence... As I'm again in the middle of another print, I'll try upgrading TabOrder tomorrow to 0.5.5 to verify that it, indeed, is not the problem.

Cheers,
F

jneilliii, 2019-11-08:
Sounds good. I also noticed you had the dashboard plugin installed.  That also has to do with temperature data and integrates with DisplayLayerProgress, so could be it as well.

closed 2019-11-29 by stale[bot]