jneilliii/OctoPrint-TabOrder #31: [Req] Drop Down for Hiding

ciordia9, 2020-02-26:
There are a number of tabs I don't care about with any regular basis (Marlin Flasher, Timelapse once Octolapse is online, Bed Visualizer, etc.) It'd be nice to be able to hide those unneeded ones in a dropdown like the default behavior for overflow is. Keep what's important in front of us.

Thanks for the plugin though. Ordering is the way. ;)

jneilliii, 2020-02-26:
There is a way to disable tabs in config.yaml documented in the appearance section [here](https://docs.octoprint.org/en/master/configuration/config_yaml.html#appearance) with some potential issues related. Your idea of moving them into a drop-down (or maybe within the built in drop-down which might be more plausible programmatically) is intriguing though. It might be fairly easy to add an option for how many tabs to show versus adding a separate drop-down, I'd have to check.

jneilliii, 2020-02-26:
So I just did a quick experiment in the developer console and this does seem to work. I would just have to add this to the on_tab_change event within the plugin to run when you change tabs.
```
$('#tabs > li:gt(2)').prependTo('#tabs > li.dropdown.pull-right.tabdrop > ul');
```
It basically moves all the tabs greater than the number specified to inside the drop-down. The only caveat I could see with this approach would be having to have the tabs you want to hide ordered to the end.  

It would be possible to take the same approach using the individual tab's id you want to selectively move into the drop-down, but that would require more programming logic and UI changes than above.  For example, to move just the timelapse tab into the drop-down.
```
$('#tabs > li#timelapse_link').prependTo('#tabs > li.dropdown.pull-right.tabdrop > ul');
```

ciordia9, 2020-02-27:
Keep it simple, a specified number and making sure you have it below the threshold is an easy win.

jneilliii, 2020-02-28:
So I did some pretty extensive testing tonight at what I thought would work, and unfortunately it doesn't.  Seems that the method used to generate that drop-down on the tabs gets bound in such a way that none of the DOM manipulation attempts I've tried are working, they keep getting moved back to the original order/functionality. I'll still try to see if there is a way to do this but right now, not so sure it's possible.

jneilliii, 2020-02-28:
I think I may have found a workaround.  Will test tonight when I'm back home and see how it goes.

jneilliii, 2020-02-29:
This was much trickier than I anticipated. In the end I had to overwrite the built-in bootstrap tabdrop plugin with another one and modify it to work with the hiding. I think it works pretty well, and adds an additional benefit of showing the selected tab from within the pull down at the top. You can install/test by using the URL below in plugin manager.  Once you've verified it's working I can make an official release.

https://github.com/jneilliii/OctoPrint-TabOrder/archive/0.5.6.zip

ciordia9, 2020-03-01:
[image]

Sorry for the delay; this is fantastic. :)

So far no conflicts with anything else and it behaves well. With them in the dropdown you can actually fit text as well as the icon. Double win.

Kudos!

jneilliii, 2020-03-01:
Cool, I'll release an official version.

closed 2020-03-01 by jneilliii