jneilliii/OctoPrint-TabOrder #34: Cannot move between hidden and assigned

tablatronix, 2020-04-29:
Maybe I missed it, but the only way to move between the 2 is to delete it and lose all settings for a tab..

jneilliii, 2020-04-29:
You should be able to drag from one to the other.

tablatronix, 2020-04-29:
Ill try again and see whats going on

jneilliii, 2020-04-30:
Ok, I see a situation where if there isn't anything in the hidden tabs section you don't have a place to drop the item from the other list. You can press the delete button on one of those higher up and then use the right [+] button to add it to the hidden section, and then you can drag others into the same area.

tablatronix, 2020-04-30:
ahh, makes sense yeah there was no way to drop there as I had no items there

jneilliii, 2020-05-16:
Just released update to set the min-height to allow dragging into empty lists.

commit 63079904 (2020-05-02): set min-height on drop areas, resolves #34
closed 2020-05-16 by jneilliii