jneilliii/OctoPrint-TabOrder #37: Not working

AtomicRas, 2020-05-25:
Can't seem to get this plugin to work. It doesn't save anything i set up.

jneilliii, 2020-05-25:
I've never heard that one before. Are there any errors in your browser's developer console? Are you forcing a refresh after update (shift+f5) in some browsers?

AtomicRas, 2020-05-25:
Sorry about that.

`Starting dependency resolution...
packed_plugins.js?afb33363:16698 [object Object]
packed_core.js?3514a079:17074 ... dependency resolution done
packed_core.js?3514a079:17384 Initial application setup done, connecting to server...
packed_core.js?3514a079:15010 Connected to the server
packed_core.js?3514a079:17349 Finalizing application startup
packed_core.js?3514a079:17224 Going to bind 48 view models...
packed_core.js?3514a079:17277 Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist
packed_core.js?3514a079:17277 Did not bind view model TelegramViewModel to target #wizard_plugin_telegram since it does not exist
packed_core.js?3514a079:17277 Did not bind view model ThespaghettidetectiveViewModel to target #wizard_plugin_thespaghettidetective since it does not exist
packed_core.js?3514a079:17277 Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist
packed_core.js?3514a079:17277 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
packed_core.js?3514a079:3334 User atom logged in
packed_core.js?3514a079:17316 ... binding done
packed_core.js?3514a079:17328 Application startup complete
DevTools failed to load SourceMap: Could not load content for chrome-extension://gighmmpiobklfepjocnamgkkbiglidom/include.preload.js.map: HTTP error: status code 404, net::ERR_UNKNOWN_URL_SCHEME
DevTools failed to load SourceMap: Could not load content for chrome-extension://gighmmpiobklfepjocnamgkkbiglidom/include.postload.js.map: HTTP error: status code 404, net::ERR_UNKNOWN_URL_SCHEME
DevTools failed to load SourceMap: Could not load content for chrome-extension://gighmmpiobklfepjocnamgkkbiglidom/include.preload.js.map: HTTP error: status code 404, net::ERR_UNKNOWN_URL_SCHEME
DevTools failed to load SourceMap: Could not load content for chrome-extension://gighmmpiobklfepjocnamgkkbiglidom/include.postload.js.map: HTTP error: status code 404, net::ERR_UNKNOWN_URL_SCHEME`

jneilliii, 2020-05-25:
Yeah, no errors there. If you change the order and save do you get the pop up message? Can you check your config.yaml file in `~/.octoprint/` folder and make sure the appearance section is changing when you save?

AtomicRas, 2020-05-25:
Sometimes when i change the order, the order in which i saved doesn't reflect however the order does change its just random.

When you "rename" a tab for example, just simply changing the first letter to a CAP, is the "main" tab suppose to stay behind in the "unassigned" tab section?

https://i.imgur.com/nvcz3vh.png

Currently printing atm, i'll check on that yaml file once its finished.

jneilliii, 2020-05-25:
Ahhh, that makes sense....the identifiers don't match to the identifiers in the unassigned so they aren't technically equal. I guess I should make that case insensitive as HTML is not and those should be considered equal.

paymon23, 2020-05-27:
I'm having a similar issue, my plug-ins aren't reordering like they're supposed too :(

jneilliii, 2020-05-27:
I need more details than what you've provided @paymon23. Can you provide a screenshot of your settings?

paymon23, 2020-05-27:
Wow so I did what you suggested above my comment, reboot the Pi and it looks to be working just fine now! Attached is a pic of my settings :)
[image]

jneilliii, 2020-05-27:
So part of the issue is that your identifiers should actually match the lowercase names as shown in the bottom list (that's why they are showing in Unassigned) and probably why icons aren't applying, etc. Realistically I should make that field read-only or just text as you shouldn't change it.

jneilliii, 2020-06-01:
Releasing version 0.5.8 which removes the ability to edit the plugin identifier.

closed 2020-06-01 by jneilliii