jneilliii/OctoPrint-TabOrder #42: [FR] optional icon color

SirFlip, 2020-07-26:
It would be usefull to make the Color optional, so it can be inherited.
Then the color changes with the text and wheather the tab is currently selected or not.
 
It's a great plugin nevertheless. Thank you

jneilliii, 2020-07-26:
Yeah, I didn't think of that since I was using the color input type, apparently their is no option for inherit. I think what I can do is split that one field up into two, and allow for manually adding the color code, which would allow removing it completely, nice idea.

jneilliii, 2020-07-26:
I've got the changes staged for the next release, but am trying to work out a reload issue with tab order settings changes prior to official release.

PR #43 (2020-07-29): 0.5.9
Allow blank color setting to inherit default behavior, resolves #42 
Added option for using the Tooltip setting as tab name, resolves #40
closed 2020-07-29 by jneilliii