jneilliii/OctoPrint-TabOrder #44: Tab Order has no effect

ranger81, 2020-08-02:
Just installed your plugin. Setting tab icons and hide tabs does work, but the sort order for the assigned tabs has no effect on my octoprint installation. I rearrange the tabs in the plugin settings, but even though I press "Save" is not properly done. Any ideas?
Thanks

jneilliii, 2020-08-02:
Are you forcing a refresh? Press either ctrl+F5 or shift+F5 in your keyboard to force a cache reload.

ranger81, 2020-08-02:
Yes, I did even clear the browser cache. The tab icon is changed properly but not the tab order.
Is there any incompatibility with Themeify?

jneilliii, 2020-08-02:
Can you please press F12 and look for errors in the developer console? It's working fine over here in Chrome. What browser are you using? I notice you are also using Themeify. Do you have the tab icon setting enabled in the plugin, could potentially cause conflict.

ranger81, 2020-08-02:
There is one error in the console, see attachment. I'm using Safari on Mac.
I have now also tried Chrome on Windows, unfortunately the same issue.
The Tab icon feature of Themeify is disabled. 

[image]

jneilliii, 2020-08-02:
Ok, the error there is not related to my plugin but it could be breaking my plugin from being able to do what it does. Not sure what plugin that might be causing that though.

jneilliii, 2020-08-02:
One thing to check would be to SSH to your pi and run the following command to read through the config.yaml and verify the appearance > order > tab section is getting updated.

```
nano ~/.octoprint/config.yaml
```

ranger81, 2020-08-02:
The config.yaml file content matches my settings in the UI.
[image]

After my print is done I'm going to disable some other plugins to see, if I can find one which causes the issue.

ranger81, 2020-08-02:
The error in the console was caused by the PrintTimeGenius plugin. Disabling the plugin did not solve my issue. However, once Octopi was restarted (because of disabling of the plugin), the tab order was correct. It seems that I need to restart the OctoPi to "refresh" the tab order, a simple force reload does not work on my installation, don't know why, but I somehow "solved" it. Thanks for your support!

closed 2020-08-02 by jneilliii