jneilliii/OctoPrint-TabOrder #47: Does not show entries In Assigned and Hidden Tabs

EddyMI3d, 2020-11-17:
In the Tab Order Settings, on some constellations, there are no entries below `Assigned Tabs` and `Hidden Tabs`
(Two installations work fine, one don't)
[Logs.zip](https://github.com/jneilliii/OctoPrint-TabOrder/files/5555773/Logs.zip)

[image]

jneilliii, 2020-11-17:
@EddyMI3d looks like your version of tab order is still not rc version. Copy/paste this link into plugin manager > get more > ...from URL and click install. After the upgrade it should work again.

```
https://github.com/jneilliii/OctoPrint-TabOrder/archive/0.5.12rc1.zip
```

EddyMI3d, 2020-11-17:
According this, yes:

[image]

At the very end of the OctoPrint log, rc1 is listed.

But I'll try installing again.

EddyMI3d, 2020-11-17:
Same result.

Do you need some data I can get via SSH?

jneilliii, 2020-11-17:
Check config.yaml to see if there are any configured in the plugin's section. Also, check the developer console in the browser for any errors.

EddyMI3d, 2020-11-18:
Ive got these entries in config.yaml

```
    taborder:
        _config_version: 5
        tabs:
        -   icon: fas fa-gamepad
            icon_color: '#00ffff'
            icon_tooltip: Control
            name: control
            showtext: false
            usetitle: true
        -   icon: fas fa-chart-line
            icon_color: '#ff0000'
            icon_tooltip: Temp
            name: temperature
            showtext: true
            usetitle: false
```
I've got no errors in the developer console.

jneilliii, 2020-11-18:
Do you have any other plugins of mine installed?

EddyMI3d, 2020-11-18:
Several:

Active Filters Extended (0.1.0)
Backup Scheduler (0.0.4)
Bed Visualizer (0.1.15)
Cura Thumbnails (0.2.0)
Dragon Order (0.1.7)
Floating Navbar (0.3.4)
M300 Player (0.1.1)
PrusaSlicer Thumbnails (0.1.2)
Stateful Sidebar (0.1.2)
Tab Order (0.5.12rc1)
Terminal Commands Extended (0.1.5)

jneilliii, 2020-11-18:
On, that's probably why. Need to install rc version of each that have a dragging sortable list. I think from your list,  blv and terminal commands extended.

EddyMI3d, 2020-11-18:
Yap - that's it. As I disabled these two, Tab Order works again in the Setup

EddyMI3d, 2020-11-18:
Installed the RCs and things run smooth 

Thank you!

jneilliii, 2020-11-18:
Thanks Eddy. I'll be releasing those versions official this weekend probably so will be less of a headache in the future.

closed 2020-11-22 by jneilliii