jneilliii/OctoPrint-TabOrder #49: Feedback and several questions

ShagoY, 2020-11-24:
Hello Jim :)

Another very interesting plugin! However, I will have some questions :

1- Are you thinking of adding the "Use Tab Name as Tooltip" feature?

2- I also use your plugin "OctoPrint-DragonOrder". Do you plan to add "navbar/sidebar" functionalities ? Just to avoid making them duplicate ?

3- With the 0.5.12, I don't have the same configuration layout. Is it normal ?
The problem is that it doesn't make the configuration simple because you don't see all the configurations I applied on each tab.
[image]

4- When the number of tabs exceeds the layout and the tabs don't have a name, the drop-drop css always uses the .dropdown-menu { min-width: 160px; } which causes this :
[image]

Thank !

jneilliii, 2020-11-24:
> 1- Are you thinking of adding the "Use Tab Name as Tooltip" feature?

[image]

The logic is a little different, You set the tooltip and have the option to either show the tab name or not, so not sure what the question is? Although, it does look like in OctoPrint 1.5.0 it's not working as expected and the Name is showing no matter what, which I'll be looking into.

> 2- I also use your plugin "OctoPrint-DragonOrder". Do you plan to add "navbar/sidebar" functionalities ? Just to avoid making them duplicate ?

I do not plan on this, they should both work without interfering with each other. Also, DargonOrder can be disabled after you modify your sidebar/navbar the way you like and the ordering will remain as that's part of the core OctoPrint functionality and that plugin just provides a UI method of doing the ordering.

> 3- With the 0.5.12, I don't have the same configuration layout. Is it normal ?

[image]

You press that button to edit your settings for each individual tab.

> 4- When the number of tabs exceeds the layout and the tabs don't have a name, the drop-drop css always uses the .dropdown-menu { min-width: 160px; } which causes this :

That's part of core OctoPrint  and there's nothing I can do to get around the collapsing of tabs into the drop-down. If you don't have any names and just have icons you'll notice that you'll see more tabs before they get collapsed.

ShagoY, 2020-11-26:
1- I understood the option "Use Tooltip as Tab Name" but I was talking about using the name of the tab as a tooltip so that I don't have to add a value in the "Tooltips" field for each tab.

2- Ok and thanks for the information.

3- This is exactly what is annoying. I would have preferred to keep the fact of having an overview of the individual options of each tab.
If icons can be problematic, putting them behind a button might be a better choice?.

4- Ok no problem and I just saw that Themeify had the same behavior.

jneilliii, 2020-11-26:
1. I got you now, make the current tab name default into that field when you add the tab to the lists, I can do that.
3. The reasoning behind changing it to a pop-up editor was to handle a memory leak that was present in the previous releases as a result of the icon picker. I was never able to figure out why but couldn't keep the memory leak and this seemed like an ok compromise.

jneilliii, 2020-11-27:
I've defaulted the tooltip text based on the tab's existing name when you add them now in the above commit. If you want to test it out you can install by copying/pasting the URL below into plugin manager > get more > ...from URL and clicking Install.

```
https://github.com/jneilliii/OctoPrint-TabOrder/archive/0.5.13rc1.zip
```

ShagoY, 2020-11-27:
Ok it works but I had to completely reconfigure the plugin because we don't have the default value of the tab for "icon_tooltip" when configuring with the old version 0.5.12.
I think you should indicate it in the readme/changelog.

But there's something that doesn't seem natural to me... I would have seen something like that:

[image]

So you keep the possibility to change the name of the tab AND the tooltip.
The "Show Custom Tooltip" checkbox can be avoided, for example if the field has a value, it displays it, otherwise not.

Also, I think that the tooltip should be enabled by default with the default value of the tab (Plugin).
This can be useful in case we customize the name of the tab, the tooltip will be a "reminder" of the plugin that is linked to the tab .

jneilliii, 2020-11-27:
I appreciate your feedback on how you think it should be and will take it into consideration. The whole point of the tooltip really was to allow people to still see some form of text when they hover over icons and are not displaying the tab name as a way of identification. I just personally can't see why you would want to have the two have different information in them and just feels like over-complication.

ShagoY, 2020-11-27:
The tooltip is just an information input that allows the user to know the reason/function/use of the button/icon/text.

In my opinion, tooltip customization is of little interest, a user leaving his mouse on an icon just to have the "pleasure" of seeing his customization are very rare :-D
On the other hand, using the tooltip to remember what the icon corresponds to is much more common ^^ ^

I left it in my sample template because you seemed to need it. 
But as you can see, I would have personally deleted it.

Also, my initial request was precisely to give the tooltip this function (to recall the original name of the tab).

Note: Im a designer UI/UX

gchedzhemov, 2021-01-10:
Hello @jneilliii,

Thank you very much for all of your hard work on the various plugins you maintain. It is greatly appreciated.

On the topic of this comment:
> Although, it does look like in OctoPrint 1.5.0 it's not working as expected and the Name is showing no matter what, which I'll be looking into.

I seem to be hitting this very same problem. Would you prefer that I open a new and separate issue for this?

Thank you again!

jneilliii, 2021-01-10:
Yes, please do @gchedzhemov I had totally forgot about that part of this.