jneilliii/OctoPrint-TabOrder #50: "Show Tab Name" toggle no longer works as expected in OctoPrint 1.5.0

gchedzhemov, 2021-01-10:
As mentioned in [Issue 49](https://github.com/jneilliii/OctoPrint-TabOrder/issues/49), the "Show Tab Name" toggle no longer works as expected in OctoPrint 1.5.0:

> Although, it does look like in OctoPrint 1.5.0 it's not working as expected and the Name is showing no matter what, which I'll be looking into.

Regardless of whether the "Show Tab Name" option is checked or unchecked, the Tab Name is always shown.

- OctoPrint version: 1.5.2
- Tab Order version: 0.5.13rc1

Please let me know if any logs, screenshots, or additional information is required.

jneilliii, 2021-01-10:
Thanks for splitting this out.

jneilliii, 2021-09-16:
@gchedzhemov is this still an issue, I can't seem to reproduce it and am wondering if there is a plugin conflict?

jneilliii, 2021-09-16:
I think I have this issue resolved with the above commit and available as version 1.0.0rc2. You can change the release channel for the plugin to `Release Candidate` in OctoPrint's software update settings.

gchedzhemov, 2021-09-17:
Thank you! Can confirm resolved with 1.0.0rc2!

[image]

jneilliii, 2021-09-17:
Great news. I may still end up tweaking the settings a little prior to release, based on some of the discussion in #49.