jneilliii/OctoPrint-TabOrder #52: Icons not workinv

DavidG-Aus, 2021-09-05:
I am probably missing something obvious but...  I just installed Tab Order and I can't get the icons to work.
Do I need to download the icons from fontawsome and if so where do I put them?

Thanks for all the work you have put into your plugins.

Cheers DG

jneilliii, 2021-09-05:
There should be a selector on that field.

DavidG-Aus, 2021-09-06:
I don't see a selector. I am using Firefox but have also tried Chrome and Edge.

[image]

jneilliii, 2021-09-06:
Something is not binding properly. Can you check your browser's developer tools on the console tab for errors on initial page load? I know firefox doesn't give as good error messages as chrome in this case.

DavidG-Aus, 2021-09-06:
I tried on chrome. There are a bunch of errors relating to MeatPack plugin but no mention of TabOrder. Are the icons supposed to be on the local machine (Pi) and if so where can I look to confirm they are there?

DavidG-Aus, 2021-09-06:
Ah, I just spotted some message relating to font-awsome
_The resource <URL> was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally.
/?#control:1 The resource http://192.168.0.70/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it has an appropriate `as` value and it is preloaded intentionally._

jneilliii, 2021-09-06:
Ah, that preload error is common, unrelated to the issue. If you disable meat pack does anything change? Force refresh your browser so cache isn't involved. Also check octoprint.log for any errors.

DavidG-Aus, 2021-09-06:
Disabling MeatPack gets rid of all the console errors but no change to TabOrder  :-(

jneilliii, 2021-09-06:
It's been a while since I've used this plugin, let me check to see what might be going on with it. What version of OctoPrint are you running and what other plugins besides MeatPack do you have installed?

jneilliii, 2021-09-06:
also, what browser? I just installed fresh on a pi zero w test OctoPrint I had lying around and it is working there so likely some other plugin conflict I suspect.

[image]

DavidG-Aus, 2021-09-06:
I am running OctoPrint V1.6.1 on OctoPi 0.18.0.  I have a bunch of plugins installed. I will try disabling them. 
BTW I have a question the Tasmota plugin - it keeps doing a power off sequence every 5 mins (idle timeout) even though the printer is already off. Is that normal?

DavidG-Aus, 2021-09-06:
Here are my plugins (couldn't find a better way to list them):
[{"key": "toptemp", "name": "Top Temp", "url": "https://github.com/LazeMSS/OctoPrint-TopTemp"}, {"key": "UltimakerFormatPackage", "name": "Cura Thumbnails", "url": "https://github.com/jneilliii/OctoPrint-UltimakerFormatPackage"}, {"key": "tasmota", "name": "Tasmota", "url": "https://github.com/jneilliii/OctoPrint-Tasmota"}, {"key": "taborder", "name": "Tab Order", "url": "https://github.com/jneilliii/OctoPrint-TabOrder"}, {"key": "GcodeEditor", "name": "GcodeEditor", "url": "https://github.com/ieatacid/OctoPrint-GcodeEditor"}, {"key": "fullscreen", "name": "Fullscreen Plugin", "url": "https://github.com/BillyBlaze/OctoPrint-FullScreen"}, {"key": "filemanager", "name": "FileManager", "url": "https://github.com/Salandora/OctoPrint-FileManager"}, {"key": "excluderegion", "name": "Exclude Region", "url": "https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin"}, {"key": "display_eta", "name": "Octoprint Display ETA", "url": "https://github.com/AlexVerrico/Octoprint-Display-ETA"}, {"key": "cancelobject", "name": "Cancel Objects", "url": "https://github.com/paukstelis/OctoPrint-Cancelobject"}, {"key": "BetterHeaterTimeout", "name": "BetterHeaterTimeout", "url": "https://github.com/tjjfvi/OctoPrint-BetterHeaterTimeout"}, {"key": "autoscroll", "name": "Autoscroll", "url": "https://github.com/MoonshineSG/OctoPrint-Autoscroll"}, {"key": "octoeverywhere", "name": "OctoEverywhere!", "url": "http://www.OctoEverywhere.com"}, {"key": "octodashcompanion", "name": "OctoDash Companion", "url": "https://github.com/jneilliii/OctoPrint-OctoDashCompanion"}, {"key": "webcamtab", "name": "WebcamTab", "url": "https://github.com/gruvin/OctoPrint-WebcamTab"}, {"key": "meatpack", "name": "MeatPack", "url": "https://github.com/scottmudge/OctoPrint-MeatPack"}]

jneilliii, 2021-09-06:
> it keeps doing a power off sequence every 5 mins (idle timeout) even though the printer is already off. Is that normal?

is that after updating just now?

DavidG-Aus, 2021-09-06:
Not sure what you mean about updating? Tasmota plugin is 1.0.4

jneilliii, 2021-09-06:
thanks, just wanted to make sure it wasn't version 1.1.0 that I just released an hour ago...

try this for me, use the plugin manager to clean up the plugin's settings with the little eraser icon and see if that changes anything. it could be settings corruption. double-check the developer console when opening the pop-up to edit the tab settings.

DavidG-Aus, 2021-09-06:
No change.  I got no console messages when I openned the TabOrder edit window.
(Just got another power off countdown)

DavidG-Aus, 2021-09-06:
I am not seeing any selector.  I am using Firefox but have also tried Chrome.

> 
> There should be a selector on that field.

jneilliii, 2021-09-06:
Only thing I can think of now is disabling all other plugins and see if it still doesn't work.

DavidG-Aus, 2021-09-07:
I disabled all my plugins and then gradually re-enabled them. Success! The plugin causing the problem is... drum roll please... OctoDash Companion.  With this plugin enabled the icons are displayed properly, I just can't select any new ones.
Is it possible to design your own icons and if so where are they stored?

jneilliii, 2021-09-07:
So with that plugin disabled the tab order icon selector works?

DavidG-Aus, 2021-09-08:
Correct.

DG

> 
> So with that plugin disabled the tab order icon selector works?
> -

-----------------------------------------------------------
  David Griffiths           ***@***.***
  Digital Graphics P/L      http://www.digitalgraphics.com.au
  Sydney, Australia         +61 2 4567 8999
-----------
Australian agents for:
 E-LAB computers - AVR Pascal and In Circuit programmers
 Logical Systems - programming adapters
 Intronix - LogicPort 34 channel Logic Analyser

jneilliii, 2021-09-16:
> Is it possible to design your own icons and if so where are they stored?

Technically, yes you could probably do a custom icon, using css and custom font, but outside of the scope of what I'm willing to implement in the plugin.

jneilliii, 2021-09-16:
I think I have this issue resolved with the above commit and available as version 1.0.0rc1. You can change the release channel for the plugin to `Release Candidate` in OctoPrint's software update settings.

jneilliii, 2025-06-12:
Oops, sometimes these things just fall through the cracks when you have 50+ plugins. Thanks for the nudge @Ajimaru will release official 1.0.0 version today.

closed 2025-06-12 by jneilliii