jneilliii/OctoPrint-TabOrder #54: Can not change the order

Guitarans, 2021-11-19:
running the latest version of octoprint.
My tab order looks like this..
[image]

it dosnt matter if i change the order in the settings..
The icons work.. even if i change the order, the tabs dosnt change possition

jneilliii, 2021-11-19:
The tabs won't re-order until you do a forced refresh (shift + f5 typically) of the browser after saving.

Guitarans, 2021-11-19:
Great, It worked.. ( the shift + F5 with my usb keyboard, But not with my laptop keyboard ) Anyway thanks for the quick reply...
i just thought something were off, since i didnt have that many options in the settings for the plugin.. Great work!!

jneilliii, 2021-11-19:
Cool and thanks for the tip.

closed 2021-11-19 by jneilliii

synthead, 2024-12-17:
Reopening this issue as I'm seeing the same problem.  This is with Octoprint 1.10.3, Tab Order 0.5.12.  I have attempted a Ctrl+Shift+R in Firefox, but the tabs remain in the same locations as before.