jneilliii/OctoPrint-TabOrder #57: cant clear icon, to have none

hayden-t, 2022-07-31:
i delete whats in the box and press close but it just comes back

jneilliii, 2022-07-31:
did you also save afterwards? Closing alone won't store the setting change.

hayden-t, 2022-08-01:
yes, and reload, but even after close if you open it back up its there again, so i guess save just saves it unchanged, thus i dont get a prompt to reload compared with when i add the icon

jneilliii, 2022-08-01:
Yeah, I've personally never tried to remove an icon before. You may have to manually do it in config.yaml until I have the time to dig into the issue further. 

SSH to your pi and run `nano ~/.octoprint/config.yaml`, use the arrow keys on your keyboard to go down to the plugins > taborder section and find where you see the icon names. Use arrow keys again to position cursor and delete or backspace to remove. Once modified press ctrl+s to save and then ctrl+x to exit. You then have to restart OctoPrint running `sudo service octoprint restart` or from the web interface's system menu in the top right by your username.

hayden-t, 2022-08-01:
a work around is to just remove the icon from assigned and add it back

closed 2022-08-01 by hayden-t